Alion3064492356/Sensors-for-RaspberryPi #5: Q: what is difference between "RGB LED" and "3colorLED?" what about their resistors±?

njackiw, 2018-05-22:
RGB_LED (original module 02) describes a 4-pin RGB LED, which in the photo looks like a conventional clear-glass bulb.

"ThreeColorLed" or "3color_LED" (original module 17) describes a 4-pin “Three-Color" LED, which looks like a surface-mount style bulb. Reading the description, it sounds exactly like the RGB_LED.

Q1. Are there any differences between the style of bulb? Is there anything to say to a new user to help them understand whether they are functionally the same, or different?

As it turns out, I was this “new user” last month. I got the Module 02-style bulb from you in a kit. I ordered some of the Module-17 SMD ones from you as standalone modules. On the Module 02-style LED, there are clearly on-chip resistors associated with each line, and the instructions don’t call for wiring up any series resistors.  On the Module 17-style one, there are no visible resistors. I wrote to you to ask if there were any there, and you said “no, you have to provide your own.” So in my use of this module, I’ve always included added resistors to keep from burning out the bulb. However, in the instructions that come with 17_3colorLed, there is no mention of adding resistors to one’s circuit. 

Q2. Does this chip (Module 17) come with series resistors to protect it against line current?   If not, should the example experiment describe adding series resistors?  What are the specs for those resistors and/or the board? (Assuming it's like the Keyes module, on Arduino, I use 180-ohm for red and 110-ohm for blue and 110-ohm for green, if the RaspberryPi output current is similar to Arduino’s.)

Please add answers as COMMENT and assign back to Nick when done.

njackiw, 2018-05-22:
If Q1 is "there is no difference except for SMD vs. long-lead-glass style bulbs," can I rename the modules to RGB_LED and RGB_SMD_LED, to suggest they are more similar?

Alion3064492356, 2018-05-23:
Thank you for your suggestion, we'd better not to rename the modules because the module's name is determined in our products list.

njackiw, 2018-05-23:
Okay, I won't rename it. But can you answer Q1 and Q2?

Q1. Is it basically the same thing, just a different (SMD) package?

Q2. Does it include series resistors or not?

Alion3064492356, 2018-05-23:
A1:There is no difference in principle between RGB LED and three-color LED, but they are different in encapsulation means.As a resullt,the code and control means are same.
A2:Since there is no resistors in our kit, our experiment does not use resistors.

njackiw, 2018-05-23:
A1. Got it, thanks!

A2. I understand the experiment does not use external resistors and there are no resistors in the kit. My question is slightly different. Are any resistors built into the Keyes-like "3ColorLed" breakout board, similar to the resistors I can see with my eye on the Kuman "RGBLED" breakout board? (Do I need to add resistors of my own, if I want to wire this into my own circuits, or are they built-in?)

For example, https://arduinomodules.info/ky-016-rgb-full-color-led-module/ is or is like the Kuman RGBLED. "This module consists of a 5mm RGB LED and three 150Ω limiting resistors to prevent burnout. "  Whereas https://arduinomodules.info/ky-009-rgb-full-color-led-smd-module/ is like the Kuman 3COLOR module, and says "use with limiting resistors to prevent burnout." So, Q2. does it include resistors in the breakout board, or no resistors on the board?

I understand the GPIO pins aren't going to source more than 16mA so the 3-ColorLED will work directly from the Pi even without extra resistors, but perhaps the description should mention these for people trying to build circuits beyond those shown in the Experiment's Wiring Diagram.

Alion3064492356, 2018-05-25:
About the resistors problem, today we decided to add it in related experiments.But we need customers to offer it themselves. This changes may cause many modifications. I will correct wiring diagrams and related descriptions, and commit to a new branch named AlionRevised2018/5/25. I feel very sorry to get you so much stuff.  Thanks again!

njackiw, 2018-05-25:
It's no problem -- I am doing this for fun, I think :-).

A couple questions/comments:
- "we need customers to offer it themselves." I don't understand this. Is it you need customers to provide their own resistors? If so, no problem, then it's like the magnet you need customers provide for hall sensors and reed sensor experiments.  
- please make sure before you branch that you've caught up with all pull requests -- there's still one open this morning (not sure why you or Emily haven't closed it)
- if you would like me to write the English part in the experiment Descriptions, no problem. Just fix the wiring diagrams and send a pull request on your branch to me; I'll modify the descriptions and you can review. I will describe how I would do this in a second comment, following.
- if you are editing the Fritzing diagrams (and therefore the *.jpg and *.cdr files) of any experiments, please add English-language labels to the sensors and ICs wired to the Pi in each diagram. If you can use exactly the same English (spelling, capitalization, etc.) as appears in the "Materials Needed" section of the experiment's *.docx, that'd be great!

Thanks Leng

njackiw, 2018-05-25:
My proposal for how to add optional, user-provided resistors to each activity that needs them
- Add a cover note in README.TXT, section 6, saying that "Some LEDs will require you to provide a 200 ohm resistor." Others (like RGB_LED) do not.

- If you're making users provide their own resistors, there are a LOT of experiments that use a generic "LED (3-pin)" for status output. Maybe you want those experiments to use the RGB_LED instead, since at least that one has resistors already built into it. I'm not sure which will be more work for you (modifying Fritzing)...probably for users, it's better if only a few experiments require them to find a resistor than many of them.

- For experiments that NEED a resistor, add "Any 150 ohm - 300 ohm resistor (you provide)" or whatever the spec is to "Materials Needed" and to step 2 of "Experimental Procedure" where it describes wiring things up per the experiment.

Please please please if you are changing any *.docx files yourself make sure your branch is up to date with all commits and pull-requests BEFORE starting your changes. Otherwise we'll have two different Word files with two different sets of changes and no way to reconcile them except redo-from-scratch.

Alion3064492356, 2018-05-26:
We prefer to use two-color led to indicate working status.For the following reasons:
1.Customer is expected to use resistors inevitably among these experiments like digital tube or two color led.
2.This experiment manual is not only for K47 kit but also for other similar kits(include resistors).
3.It is possible to add resistors in this kit some day, although it's not up to me.

Alion3064492356, 2018-05-26:
I'm not clear about the problem you mentioned above that "Please please please if you are changing any *.docx files yourself make sure your branch is up to date with all commits and pull-requests BEFORE starting your changes. Otherwise we'll have two different Word files with two different sets of changes and no way to reconcile them except redo-from-scratch.". I have no idea about how to operate. May I send the revised files to you through email?

njackiw, 2018-05-26:
You are using Git, right? So this morning (after the e-mail above), Emily merged #20, which means those changes are now in the "master" branch. If you make a branch from master now, you'll start with those changes already in place. If you already branched earlier, before she merged, you will need to "merge master's changes into [your branch]" before  you can commit your changes back into master if there are any conflicts. If there are no conflicts (if you are working on different files than I was) it will handle the merge automatically, no problem!

I'm not an expert in Git, so maybe you can find a clearer explanation online :-).

njackiw, 2018-05-28:
I am still unclear on the status of plans here.

My understanding is that several of the Kuman LED units will basically burnup if I connect them as in original wiring diagrams. They need resistors but do not include resistors, and the wiring diagram shows no resistors, and the instructions make no mention of resistors. I think this is true of all LED modules except RGB_LED bulb, which contains built-in series resistors protecting the LED.

So, here are some options about what we could do:

Here are some options I can think of:

1. provide built-in resistors to all our modules — unrealistic, since this would require manufacturing change

2. provide a few cheap standalone resistors and modify (many) experiments to describe how to connect them to our LEDs — would change packaging, and (somewhat) product cost, also lots of determining technical requirements and work revising wiring diagrams and activity write ups, etc.

3. don’t provide cheap resistors but modify experiments to say “YOU provide your own resistors” and document required resistors where they are required — make users provide them. No change to packaging but still considerable documentation work, revising descriptions and wiring diagrams.

4. continue to stay silent about the fact that our instructions are wrong and will lead to the destruction of our components if you follow them.

Has Kuman already decided which of these we WILL do in this modification? 

Of course since you’ve asked me to help make things more understandable and better for your customers, I do not like answer (4). But it is not up to me. What answer does Kuman decide on? If you cannot do (1) or (2) for cost/manufacturing issues, but decide on (3), I am happy to keep working to make the documentation better in this regard (i.e. documenting resistors as appropriate), but Leng will have to do the work of updating wiring diagrams, and I will also need to know WHAT SPECIFIC TYPES (ohm-ratings) OF RESISTORS are needed by each of your various LED modules.

njackiw, 2018-05-31:
Just coming back to this to see if we can close it. I think the history is:

- 5/28 Emily wrote "As for #5,we chose solution 3 and let's work on this."
- 5/29 you decided to switch to #2 instead
- Leng fixed most activities to describe use of resistors in **AlionRevised2018/5/29** branch
- Nick fixed copy-edited that and fixed remaining issues in #35 
- All of the resistiors work is done and 3ColorLED now has a better name too (RGB_SMD_LED) and has been renamed to both its better name and its conventional name

So I think there is no more issue here. If you agree, please CLOSE this issue.

njackiw, 2018-06-04:
No one has disagreed with what I wrote 4 days ago here, so I'm going to close this one myself.

closed 2018-06-04 by njackiw