Alion3064492356/Sensors-for-RaspberryPi #6: Q: Which LED should shock/tilt/knock switch Experiments use?

njackiw, 2018-05-22:
Shock Switch (original module 04) and many like it (Tilt, Knock, etc.) describes wiring that sensor to  a 3-pin status LED with only the signal and ground LED pins connected. 

The kit contains many LEDs that have different "module names" and so new users will be confused if you just mention "an LED". Which specific LED do you recommend they use±?

2colorLED?
7colorLed (original module 16)?

Please add answer as COMMENT and assign back to Nick for action.

Alion3064492356, 2018-05-23:
In the Shock Switch experiment, we use the two-color LED to indicate the working status about shock switch.The two-color LED has three pins  representing green led, red led and GND respectively,we only connect pin GND and pin Green because we just need these two pins,and the pin Red should be connected to nothing.
As for your second question "The kit contains many LEDs that have different "module names" and so new users will be confused if you just mention "an LED". Which specific LED do you recommend they use±?",we use two-color LED,could you please modify it?

njackiw, 2018-05-25:
See #5  too. Perhaps if users have to provide their own resistor for **2ColorLED**, then it might be better if other experiments (like these) used **RGB_LED** instead? (Just because it doesn't require resistors?)

I'm not sure which would be better.

njackiw, 2018-05-31:
Closing because we now have resistors for all our LEDs and can use **2colorLED** here.

closed 2018-06-01 by njackiw