Alion3064492356/Sensors-for-RaspberryPi #7: Q: What is the correct pinout of the 2color LED?

njackiw, 2018-05-22:
Your two-color LED modules have three pins, “-“ on the left, {blank} in the middle, “S” on the right. 

The question is: how do {blank} and “S” map to RED and GREEN?

The original pin diagram text file says that “S” and “G” are different:

	"S" =====> 10
	pin_G =====> 11

suggesting that {S->RED} and {blank->GREEN}. This is option 1.

But the labeled picture of the module in your Word file says that “S” and G” are the same pin!

[image]

This suggests that {S->GREEN} and {blank->RED}. So that’s option 2.  

Finally if I look at the wiring diagram and Raspberry Pi pinout

[image]

 it shows that {blank -> 10}, {S -> 11}, where the Python code says

	pins = {'pin_R':10, 'pin_G':11}  # pins is a dict

so therefore {blank -> R,  S -> green} … which is Option 2 again.

So is the pin diagram wrong, or the illustration?  Please confirm with a correct statement about the pins on the module, and I will fix the illustration, pin out diagram, and code to match.

Alion3064492356, 2018-05-23:
Sorry to hear that, It is my mistake.The correct wiring means is as the wiring diagram shows. The wiring description is wrong in the Pin position corresponding table.txt.

njackiw, 2018-05-23:
Thanks; I will fix it in text.  Reassigning defect to me until I commit the fix.

closed 2018-05-24 by Emily2333