Alion3064492356/Sensors-for-RaspberryPi #12: Q: Also confused about REED SWITCH sensor

njackiw, 2018-05-23:
This relates somewhat to my confusion about the various Hall sensors.

Kuman's REED SWITCH sensor has a threshold-adjustment potentiometer, and the EXPERIMENTAL code that goes with it includes both an analog (A0) and a digital (D0) readout of the sensor.

But everything I read about reed switches, including the "technical introduction" in your EXPERIMENT, suggests they are electro-mechanical switches that are either open or closed -- wire conducts or it doesn't. That corresponds clearly to a digital readout of OPEN or CLOSED, HIGH or LOW. There is no "proportional voltage" in such a switch.

So what is the analog measurement measuring? What is the threshold potentiometer and linear comparator comparing against??

I can find no information on the internet that suggests that "thresholds" are relevant to reed-switch technology...

njackiw, 2018-05-23:
(In contrast, the mini-reed switch makes perfect sense, and has no analog component.)

Alion3064492356, 2018-05-24:
As for the reed-switch, It has two signal pins named AO,DO respectively.You can only use DO to control LED     directly and ignore the pin AO. However, We use both in order to show you how to use the pin AO although it is not necessary.

Alion3064492356, 2018-05-28:
Sorry to confuse you! I reviewed it and found it's my fault! What I make sure is pin AO is useless in this module .We should connect pin AO to nothing because this sensor is digital, and I am gonna revise this experiment and commit it.

njackiw, 2018-05-28:
See #28 and #29 for a proposed module renaming that better explains how this product relates to other reed sensor products, using the same language as used when explaining differences between various Hall sensors and various temperature sensors.

njackiw, 2018-05-28:
Okay you and I are on the same page -- these are pretty confusing and it's hard to tell why anyone would think they're useful -- but I finally came up some ideas about why they might be. Please see if you like my explanation in #28 that begins "However, one way that I _can_ make sense...". Applies to miniReed and magneticHall too!

njackiw, 2018-05-28:
sorry, I deleted last comment. I'm wrong to put miniReed in with magneticHall and digitalTempSensor as "analog with convenient digital" -- it really _is_ digital. Your last comment is right -- go ahead and revise :-)!

closed 2018-06-08 by Alion3064492356