Alion3064492356/Sensors-for-RaspberryPi #13: Nick's proposals for renaming and reordering things

njackiw, 2018-05-23:
Nick would like to make some changes to the sensor folder order and document names to improve clarity and easy of use.  Please discuss and add COMMENTS to this issue letting me know whether Kuman is okay with me making these changes. (You can approve or reject changes individually.) I would like to make them all in one pass, near the end of the process, so I can update various pieces that depend on them at the same time.

A. Rename **3ColorLED** to either **RGB_SMD_LED** or **miniRGB_LED**.

Reason: consistency. The **3ColorLED** is the same thing as the **RGB_LED**, just smaller. It is NOT like the **2ColorLED** or **7ColorLED** -- which meaningfully produce only two or seven colors --- instead it is meant to produce many colors out of R, G, B combinations. Renaming it to **RGB_SMD_LED** correctly describes it as _like_ the **RGB_LED** but different by virtue of SMD packaging; renaming it to **miniRGB_LED** correctly describes it as _like_ the **RBG_LED** but different by virtue of being smaller. (Both are true. We elsewhere already use the **XXX** and **miniXXX** naming convention with reed switches.)
 
B. Rename **Read Me First.xt** as **00_ReadMe.txt**.

Rationale: this way it will appear before all sensor folders in any alphabetized list.

Afterwards, scan documents to make sure no one mentions **ReadMeFirst.txt**

C. Exchange **reedSwitch** with **miniReedSwitch** in numeric folder order. The mini-reed switch is the easiest (most understandable) magnetic sensor to start with; users should encounter it before the non-mini one.

Alion3064492356, 2018-05-24:
A. Okay, I agree to rename 3ColorLED to RGB_SMD_LED, but I think it is better to add (3colorLed) after  RGB_SMD_LED like RGB_SMD_LED(3colorLed),  for the reason that this module's name is 3colorLed in our product list, and It will not be modified for the moment. 
B. It's ok to rename Read Me First.xt as 00_ReadMe.txt.
C.It's better to exchange  reedSwitch with miniReedSwitch in numeric folder order.
D.I think we could locate  analogTempSensor, digitalTempSensor, ds18b20 and DHT11 together and put them to the end for the reason that ds18b20 and DHT11 are much hard to learn.
E.We could rename DHT11 to tempHumiditySensor as the way to rename 3ColorLED .
F.I absolutely agree your suggestions.

njackiw, 2018-05-28:
Reassigned to Nick until he actually makes these changes, but no further questions on this.

njackiw, 2018-06-04:
**Nick's Proposed Ordering & Grouping**

Separate from ReadMe.txt I am hoping to have an Overview or Introduction-to-Sensors Word file designed for the new user. In this document, I have several goals, including introducing each of the sensors very briefly (one line description), and presenting them in "groups" of related functionality. I propose that folder-order in the file directory follow this same order, i.e. that are folders are numbered so that all experiments within a group appear numerically together, and that the numerical series within a single group should move from the "conceptually simplest" sensor in that group to that weirdest, most complicated, or least useful. So in this proposal "by group" is the major ordering principle, and "by complexity" is the minor ordering principle.

This document is also the place where I will address two other important points for thinking about the collection of sensors:

- The difference between "conventional names" and "better names." Thanks to Leng's renaming, many of our experiments now have better names by default. There are still 5 or 6 that have bad conventional names, however, where a better name would be clearer.  If there's any chance we could still rename any of these in our own materials, now is the time to decide that! The worst two conventional names are the "digital tubes" (8-segment-displays) -- see my comments above. The Hall Sensors are also poorly named, conventionally.  With the reed switches, it's not the conventional names as so bad, the problem is that one of the sensors itself doesn't make good sense (#12) and so no name is "sensible!"

- The differences between "analog", "digital", and "hybrid sensors." Since first meeting the third category in your collection, I've learned there is a standard engineering term for it -- "mixed signal." Unfortunately this is not a good name to use for beginners, because outside of engineering "mixed signal" in English means ... ha ha ... "confusing!" ("Stop giving me _mixed signals!_ I can't understand you!!")  So I think we are best sticking with "hybrid" here. This only affects one or two sensor name, so not a big deal, but I want to be able to give new users a "big picture" introduction to these three different types of packages, since almost all of your sensors appear in one or the other form.

Before I write this Overview, I want to settle on the groups and ordering principle. The following table has my proposal.     Please pay most attention to the left two columns: the first is the group, the second is the sensor. So the second column is in the same order I propose we use for our folders. When reviewing this proposal, please ask yourself the following questions

Q1.  do these groups make sense? 
Q2. does each sensor belong in the group Nick's put it in? 
Q3. does the order through sensors within a specific group make sense, as a progression from "easy" to "hard?"

If the answer is ever **no** please add a comment here! I'd like to try to finalize this quick so I can rename the folders and write the Overview. 

As you come to understand the group concept, I have another question about actual folder names on disk. Right now our collection consists of thirty-eight (or 37) different experiment folders, named like this:

- 01_experiment1
- 02_experiment2
- ...
- 37_experiment37

We could keep this structure using the order in column 2 of the table below, or we could introduce "group folders" at the top level, like this:
- 1 Sound and Light Displays
- - 01_2colorLED
- - 02_RGB_LED
- - ...
- - 09_4bit_digital_tube
- 2 Light and Infrared Sensors
- - 10_photoResistor
- - 11_infraredEmitter
- - ...
- - 16_trackingSensor
- 3 Touch and Impact Sensors
- - ...
- ...
- 7 Miscellaneous
- - ...

So, the remaining question is, 

Q4. Should we introduce Group Folders like in the above? Or keep it one flat list of folders for all sensors?

Here is the proposed ordering, for your consideration.

sound & light displays (outputs) |   |   |  
-- | -- | -- | --
  | 2colorLED |   | output
  | RGB_LED |   | output
  | RGB_SMD_LED(3colorLED) |   | output
  | 7colorLED |   | output
  | activeBuzzer |   | output
  | passiveBuzzer |   | output
  | laser |   | output
  | 8segment_digital_tube | 8digitLEDdisplay | output
  | 4bit_digital_tube | 1digitLEDdisplay | output
light & infrared sensors |   |   |  
  | photoResistor |   | analog input
  | infraredEmitter |   | output
  | infraredReceiver |   | digital input
  | lightBlock |   | digital input
  | flameSensor |   | hybrid input
  | obstacleAvoidance |   | digital input
  | trackingSensor |   | digital input
touch & impact sensors |   |   |  
  | buttonSwitch |   | digital input
  | touchSensor |   | digital input
  | tiltSwitch |   | digital input
  | metalTouchSensor |   | hybrid input
  | knockSwitch |   | digital input
  | shockSwitch |   | digital input
  | joystick |   | analog input (x + y) + digital input (button)
sound sensors |   |   |  
  | bigSoundSensor |   | hybrid input
  | smallSoundSensor |   | hybrid input
magnetic sensors |   |   |  
  | miniReedSwitch | basicReedSwitch | digital input
  | reedSwitch | ???nonsenseSensor??? | hybrid input
  | analogHall | analogHall | analog input
  | HallMagnetic | digitalHall | digital input
  | linearHall | hybridHall | hybrid input
temperature sensors |   |   |  
  | analogTempSensor |   | analog input
  | hybridTempSensor |   | hybrid input
  | digitalTempSensor(DS18B20) |   | digital input
  | tempHumiditySensor(DHT11) |   | digital input
  |   |   |  
miscellany |   |   |  
  | smokeSensor |   | digital input
  | rotaryEncoder |   | digital input
  | relay |   | controller

Alion3064492356, 2018-06-06:
Thank you, Nick!
First of all, I completely agree to regroup and sort folders.
Secondly, the  classification list you provided is exciting.I suggest to put this table below the top level folder, then the corresponding information of the old and new file names in the table.
Thirdly, There is a type error in the new names of both 4bit_digital_tube and 8segment_digital_tube. And I suggest not to change reedSwitch because we have no idea about its new name.
Finally, we could rename these experiment but we have to keep their old name in this table.

njackiw, 2018-06-06:
Hi Leng!

>First of all, I completely agree to regroup and sort folders.

Okay, will do this today.

> Secondly, the classification list you provided is exciting.I suggest to put this table below the top level folder, then the corresponding information of the old and new file names in the table.

Okay, will do this, including the table in my Introduction doc.

> Thirdly, There is a type error in the new names of both 4bit_digital_tube and 8segment_digital_tube. 

Fixed (in the table above, which should now be correct).

> And I suggest not to change reedSwitch because we have no idea about its new name.

Okay, so we'll keep the "conventional" names here: "miniReedSwitch" (the sensible one) and "reed switch" (the not-so-sensible one :-)).

> Finally, we could rename these experiment but we have to keep their old name in this table.

Okay, really?  I had thought we'd decided the opposite earlier (don't rename experiments, but put NEW name in their Description DOCX). I'm happy to rename and put old names in table AND in Description DOCX, but please **confirm** I am understanding you correctly.  If so, I will rename folders, files, and contents of Description to use the six "better names" listed in the table above, while making sure that those six  experiments mention their "conventional name" both in the Overview table and in the experiment-specific Description.

Alion3064492356, 2018-06-08:
Yes, we can rename modules. However, one principle should be adhered to that only the very unreasonable name should be modified and then be mapped in the table.
Thanks!

njackiw, 2018-06-10:
Okay, introduction_to_sensors.docx now has proposed text for the introduction. Please read and send me your comments! 

Also, pull request #58 takes care of "Yes, we can rename modules. However ... only the very unreasonable name should be modified and then be mapped in the table." That is done, and the unreasonable name is also mapped in the .docx file for those (now reasonably renamed) experiments.

I think this Issue can be closed if you both are relatively happy with introduction_to_sensors.docx.

njackiw, 2018-06-13:
introduction_to_sensors has been reviewed and feedback incorporated into a final version. We can close this one!

closed 2018-06-13 by njackiw