Alion3064492356/Sensors-for-RaspberryPi #14: Proposal: replace Infrared Emission & infrared Receiver experiments with one experiment

njackiw, 2018-05-23:
This is a big proposal, but Emily has asked me to help make these experiments make good sense to an English user, and I believe these two experiments don't make sense to any user in their current form.

One experiment turns on an infrared (IR) lightbulb. Unfortunately, no one can see infrared so this has no experimental effect. So at the same time, it turns on a regular lightbulb, which can be seen. But that lightbulb has nothing to do with the infrared emitter!

The other experiment detects infrared signals. Unfortunately, it doesn't give users any way to create infrared signals, so nothing understandable happens.

I suggest replacing these two experiments with one experiment, called **Infrared**.  If you need to have two folders so you can count this as two different sensors, put the SAME experiment into two folders named **infraredEmitter** and **infraredReceiver**....they would be identical.

The purpose of the experiment would be to send infrared signals and detect them, and would look like this (pseudocode)

```
var lastEmittedSignalTime = 0;
BOOL lastEmittedSignalState = true;

var lastDetectedSignalState = false;

while (1)
{
         // DETECTION: see if we have started, or stopped, detecting infrared; and
        // report changes in what we previously detected to the user.

        if (lastDetectedSignalState != getDetectorState())
       {
             lastDetectedSignalState = not lastDetectedSignalState;
             if (lastDetectedSignalState)
                     print "Detector sees the light go on!";
             else
                     print "Detector sees the light go off!";
       }

      // EMITTER. Unrelated to the detector, turn the emitter ON or OFF at one
     // second intervals

         if (currentTime() > lastEmittedSignalTime + 1 second)
         {

                  lastEmittedSignalState = not lastEmittedSignalState
                  setInfraredLEDState(lastEmittedSignalState)

                  if (lastEmittedSignalState)
                        print("Turning emitter ON"
                 else 
                        print("Turning emitter OFF"

                  lastEmittedSignalTime=currentTime()

        }

```

I can't write and test this, because I don't have the sensors or a Pi, but this code would be the basis of a demonstration that shows that both the emitter and detector actually work and can be used to signal information between them --- even though one cannot see that information with the human eye.

What do you think? If you agree, can Leng put together this experiment easily?

Alion3064492356, 2018-05-24:
I agree to modify these two experiments and I will deal with it. 
If you have an android phone like samsung or huawei ,you could see the infrared emission with your phone camera.

njackiw, 2018-05-24:
Your new code is great -- much clear. Thank you!  In a coming pull request, you'll see I have made a couple important modifications/extensions:

a) I rewrote the English in the Description.docx to be more explanatory of the overall methods
b) I added some comments to both the C and Python source to explain the interrupt handling logic
c) I revised the filenames of corresponding (identical) in both folders to be the same, in all cases  simply "infrared.*" -- infrared.jpg, infrared.docx, infrared.c, etc. The folder names remain different (emitter vs. receiver), and the Description makes it clear (as the filenames attempt to) that there is only _one_ experiment for _both_ sensors.

This is all more work (and takes more space to explain) than most activities, but there's a lot going on here. We are using three sensors at once (including the LED), two of them new; we are using interrupt routines critically for the first time in any experiment; and we are sensing a physical phenomenon that is invisible to humans! I think there are plenty of opportunities for users to be confused about one or more of these details, so I hope that our extra attention here is worthwhile.

Leng, please review these changes carefully. If you don't like the way I've explained or organized things, let me know and we can revert to your last commit and try something in a different direction.

Otherwise (if you like them as checked in), you can CLOSE this comment.

Thanks
Nick

Alion3064492356, 2018-05-25:
I review your changes and I feel it is great.

closed 2018-05-25 by njackiw