Alion3064492356/Sensors-for-RaspberryPi #16: Proposal: promote activeBuzzer and passiveBuzzer into their own top-level folders

njackiw, 2018-05-23:
Right now K47 has two different hardware sensors, with two different sets of sample code and two different activities, buried in one folder called **05_buzzer**. Why? You don't bury all of your different LEDs in one folder, or all of your different temperature sensors, etc.

To me, one folder makes sense for the 3mm vs. 5mm **2ColorLED**, where the only difference between the hardware units is size, but the experiments, and user understandings, of the components is essentially identical.  (I have collapsed the two different **2ColorLED** documents into one, based on this identity.) But the buzzers are very different from each other.

Can we separate these into two top-level folders **xx_activeBuzzer** and **yy_passiveBuzzer**? This may impact how many components you "advertise" in the collection, as there will be 37 folders rather than 36....

Alion3064492356, 2018-05-24:
Okay, as  you say, we should seperate these two buzzers into two top-level folders. Thanks a lot.

njackiw, 2018-05-24:
Okay, reassigning to this issue to me for my editorial action

njackiw, 2018-05-24:
done in Pull Request #20

closed 2018-05-24 by njackiw