Alion3064492356/Sensors-for-RaspberryPi #17: Proposal: simplify Button experiment code

njackiw, 2018-05-23:
The **Button** should be one of your simplest sensors to understand, and early place for your customers to feel like they have success with your products. However, in its original form, the source code that accompanies **Button** experiment is more complicated than it needs to be. The code changes the light every time the button is pressed, meaning that the code needs to debounce button presses, track previous light state, and wait for button-releases. This means it has an inner loop, conditional states, etc. Could we simplify it to

```
	while(1)
	{
              // when the button is pressed (LOW), turn the light on;
             //  when the button is up (HIGH), turn the light off.
            digitalWrite(LedPin, !digitalRead(BtnPin));
       }

```
This seems to me simpler and clearer than the current code.

njackiw, 2018-05-24:
Thanks for the comments. Based on them, I looked at button_Int.c, which is not mentioned in the Description.docx file at all. I like that it is simpler than the code that is in the Description.docx file (and in button.c, which is the code I complain about above), but I still think it is not simple code. If like me a customer is new to Raspberry Pi, he may not know in this button_int.c code what ISR means, how interrupts work, or why we detect only on the INT_EDGE_FALLING and not on rising or other conditions.  And none of these details have anything specific to do with buttons!

Personally, I think code like the pseudo-code I show above is about as simple as it gets: the light is on when digitalRead(BtnPin) is off; the light is off when digitalRead(BtnPin) is on.

I would prefer that code, but for the time being will keep the code in Button.c and in the *.docx file as is.  You and Emily can decide at your leisure if you want to simplify it to the example above, or have me add comments to the .docx file explaining what Button_Int.c is doing (in addition to Button.c). I think either would be an improvement to the situation we have now, where the main (.docx) code is complicated, and the Button_int.c code is more complicated than it needs be and also unmentioned in the documentation.

Thanks,
Nick

Alion3064492356, 2018-05-25:
Thanks for your advice. It is probably simpler to those customer is new to Raspberry Pi about your code. However, I would like to remain button_int.c and button.c because we use pull-up resistor in our experiment.  
For easier to understand, I am going to add some annotation in code.

njackiw, 2018-05-25:
okay, sounds good to me. Please base your changes on current code -- open pull request #20 already has some commits to the Buttons module.

Alion3064492356, 2018-05-26:
Okay, I am going to revise this module as soon as possible.

closed 2018-06-25 by Alion3064492356