Alion3064492356/Sensors-for-RaspberryPi #18: Seeking clarification about C vs. Python pin outs

njackiw, 2018-05-23:
Most experiments' C source code uses different pin assignments than either Python source code or the experiment's **Pin position corresponding table.txt.** As but one example, consider **button**:

Pin position corresponding table.txt:
```
Button Switch pin position:
	"S"  =====> Raspberry Pi pin 11
...	
LED pin position:
	"S" ======> Raspberry Pi pin 16
```

Python:
```
BtnPin = 11
LedPin = 16
```

C source code:
```
#define BtnPin    0
#define LedPin    4
```

In general, the **Pin position corresponding table.txt**, Python source code, and Wiring Diagram usually agree on pin assignments, and the C source code uses completely different pin numbers.

Q: is this a difference between what C calls the same physical pins and what everything else does, or is this a problem with the C source?

If it's a problem with the C source, I can fix it where C disagrees with everything else.

If it's no problem -- i.e. the C source's weird pin numbers because pins are numbered differently in C than they are on the board (or in Python) -- maybe we should make a note in the ReadMe about this. As a new user to Raspberry Pi but someone with a fair amount of coding experience, this looks to me like a bug, so if I have a faulty understanding the product could help educate me to a correct one.

Alion3064492356, 2018-05-24:
It's no problem.RaspberryPi  has three ways to encode pin number, BOARD encoder means pin number is same as the RaspberryPi board physical pin number, BCM encoder means pin number is as BCM encoder specified and wiringPi encoder means pin number is as wiringPi encoder specified.You could search it on the Internet.

Alion3064492356, 2018-05-24:
The python code use BOARD encoder and the C code use wiringPi encoder.

njackiw, 2018-05-24:
Thanks for the clarifications! Makes sense, and I will not worry about C pin assignments then :-). Closing this issue.

closed 2018-05-24 by njackiw