Alion3064492356/Sensors-for-RaspberryPi #9: Q: AnalogHall/Voltage Difference?

njackiw, 2018-05-23:
The original analogHall Description says

> A voltage difference is generated when the Hall element and the magnet meet in the forward direction, and there is no voltage difference when the Hall element and the magnet meet in the forward direction.

This is contradictory. Either there IS a voltage difference in the forward direction or there IS NO voltage difference in the forward direction, but it can't be both.  I think maybe one time you mean “backward” and the other time “forward” but can’t quickly decide which one is which! What is correct here?

Please answer and assign back to Nick for correction in the Word file.

Alion3064492356, 2018-05-24:
A voltage difference is generated when the Hall element and the magnet meet in the forward direction, and there is no voltage difference when the Hall element and the magnet meet in the inverse direction.

njackiw, 2018-05-24:
Excellent; I will fix in a coming commit. Thanks.

closed 2018-05-24 by njackiw