Alion3064492356/Sensors-for-RaspberryPi #21: Laser: help me understand

njackiw, 2018-05-25:
I'm confused. Does the laser produce light visible to the eye, or not?

The original Laser experiment begins:

> This course will .. control the laser transmitter to emit laser signals, and show the working status of the infrared emission tube by the LED light on and off.

I'm not sure what "infrared emission tube" has to do with it (unless the laser is sub-visible), or why we need an LED. Can you explain? Maybe the laser produces only invisible (IR) light, in which case the original Experiment's design is fine.  Or maybe the laser produces visible light, like a KY-008, in which case the original Experiment needs some language revision and could be simplified to not include the LED (which is a redundant visible light -- it wouldn't add anything to the experiment if the laser itself is visible!).

Your clarification welcome!

Alion3064492356, 2018-05-26:
Our error! "infrared emission tube" is an error, we should correct it. Laser is visible and may bring dmage to your eyes when you are looking straight at it, so we use a led to indicate it's working status.

njackiw, 2018-05-27:
Okay, I will correct. Reassigned to me.

njackiw, 2018-05-27:
Closed in #23

closed 2018-05-27 by njackiw