Alion3064492356/Sensors-for-RaspberryPi #22: Tilt Switch: Help me understand difference to ShockSwitch

njackiw, 2018-05-25:
**ShockSwitch** originally said "SW-520D is a ball type tilt-induced vibration unidirectional trigger switch"

How is **TiltSwitch** any different? What can I say to a user that helps them understand the difference? Is **TiltSwitch** a conventional mercury switch (like KY-017)?

I don't have the sensors to play with, so if there's anything you can say qualitatively about the difference in "what sort of tilt" they measure, that would be great to know.  For example, maybe you really have to shake the ShockSwitch hard, but the tilt sensor detects more gradual changes? (I'm making this up; I don't know how they behave!)

Help!

Alion3064492356, 2018-05-26:
In our old kit, TiliSwitch is mercury switch. we will ship this kit to Europe with the development.But it is a problem for this kit that mercury is forbidden to be shipped to Europe,as a result, we replaced it with ball type tiltSwitch.
About "ShockSwitch originally said "SW-520D is a ball type tilt-induced vibration unidirectional trigger switch",I think we could delete "tilt-induced".what do you think?
The difference is ShockSwitch detects vibration but TiltSwitch detects tilt.

njackiw, 2018-05-27:
Okay, sounds good. I will modify ShockSwitch and revise TiltSwitch to match.  Reassigning this Issue to me for action...

njackiw, 2018-06-01:
Fixed in #44

closed 2018-06-01 by njackiw