Alion3064492356/Sensors-for-RaspberryPi #25: Q: Analog Temperature Sensor/how to make LED go on?

njackiw, 2018-05-27:
In this temperature sensor experiment, the original **Description.docx** says only:

> Experimental phenomena
> When the analog temperature sensor's analog output value meets the threshold condition, the LED will light on; when the threshold condition is not satisfied, the LED light goes out.

Q: Can you give me an example of how the user should heat the sensor to make the LED come on? I imagine the experiment is calibrated so at "room temperature" the LED is off. Is it sufficient to squeeze the thermistor with one's fingers (LED turns on at body temperature)? Should they hold a flame to it? Since I don't have access to the actual thermistor or running code, give me some sense of a suitable procedure that will generate enough heat to flip the LED.

Alion3064492356, 2018-05-28:
In this experiment, we should need a flame to trigger this sensor but be careful to avoid too close to burn sensor out. If you want to trigger this sensor with your fingers, you're expected to set threshold much larger such as 225 or others.

njackiw, 2018-05-28:
Got it; reassigned to me til fixed.

njackiw, 2018-05-31:
Fixed in #35

closed 2018-05-31 by njackiw