Alion3064492356/Sensors-for-RaspberryPi #26: Q: TouchSensor/relation of actual sensor to "liquid crystal" description

njackiw, 2018-05-28:
In **TouchSensor.docx**,  you say:

> [Touch sensor] directly replaces traditional switches, providing comfortable operation, excellent feel, precise control and no mechanical wear. At the same time, the touch switch is more humane, you can choose the text prompts on the switch, personalized text labels show the effect of liquid crystal display. Crystal panel issued a faint glimmer, which make late night is no longer completely dark, enough to make people form a sense of position and contour. 

When you mention "text prompts" and "faint glimmer," are you talking about properties of the actual sensor that is in the kit? Or are you discussing some touch sensor applications, where a touch sensor (like in the kit) has been embedded above (or below?) a liquid crystal display? 

(Remember I don't have access to the sensor itself to examine or test.) The current text makes it sound like the sensor itself has "text prompts" and possibly a glimmer, but I see no further reference to how to access those properties of the sensor.

Alion3064492356, 2018-05-28:
This overview is talking about some touch sensor applications,  It points to a product using touch sensor module and liquid crystal display. Touch sensor is a part of it.

njackiw, 2018-05-28:
Okay, got it.  Will revise; assigning to me for action.