Alion3064492356/Sensors-for-RaspberryPi #27: Questions/help me understand 4bit_digital_tube

njackiw, 2018-05-28:
**4bit_digital_tube.docx** gives a 12-pin pinout diagram including pins [1, 2, 3, 4], [a, b, c, d, e, f, g], and [dp] ...  This pin description does not include GND.  By contrast, **Pin position corresponding table.txt** describes a 13-pin pinout, including GND.

[image]
 
Q1. How many pins are there really, and what is their order on the board? (If they are not labeled on the module, we need an accurate pinout illustration.)

More importantly than this, neither of the above pinouts give a description these pins' **purpose** have or how they relate to the 32 separate LEDs on the display (8-segments-per-digit x 4-digits = 32 segments).

Q2. How do the segments of the display get controlled by the pin diagram?

Reading the code, I get an uncertain sense that I can write a single digit-value into any of the four digits-columns by setting the column's [1..4] pin HIGH and the digit-value's segments [a..g + dp] LOW.  This works but only describes how to set one digitColumn on the display. If I want to set multiple digitColumns -- e.g. display "1234" all at once -- how do I configure this pinwise? Is there some sort of serial logic in effect -- the digits scroll left and I only control the rightmost digit?

Alion3064492356, 2018-05-28:
Q1: It have 12 pins totally. pin GND is an error, should be deleted.
Q2:My explaination might be fuzzy becasue my poor English. Even so, I will do my best to explain this question.This module consists of four digital tubes. Each tube consists of 8 leds including decimal point,and be controled by 9 pins (one is bit-select pin like 1 , 2, 3 ,4 and others are named a, b, c, d, e, f, g).these pins named after letter of one tube are in parallel with another tube.we could control a tube through it's bit-select pin.

njackiw, 2018-05-28:
I still don't understand Q2 even with your answer. My question is about setting displaying multiple digits with different digit values at the same time.

Let's say I want to set "89" in the first two digits at once. Reading SevenSeg.cpp (Arduino library for this sort of device), it looks like I this sequence of events to get the "8" in the first digit":

(a) erase all the segments by setting segment pins OFF, 
(b) bit-select the 1st digit pin by setting its pin ON and the other three digit pins OFF 
(c) set the segment pins to the shape of "8" which I want as the first digit,

Then I do a similar thing for "9": 
(d) erase all the segments by setting segment pins OFF, 
(e) bit-select the 2st digit pin by setting its pin ON and the other three digit pins OFF 
(f) set the segment pins to the shape of "9" which I want as the second digit
etc.

**But!** What I don't understand is: on what condition or signal does the tube actually update the display? If it happens to the currently-bitselected digit every time I change a segment pin, then it happens during steps (a), (c), (d) and (e) above. But step (d) erases all the first-digit display it just set in step (c) before we move on to digit 2!  So the final display of the first digit is erased, and our two-digit output is just " 9" rather than "89"!

On the other hand, if it only happens when every time I change the bitselect pins, then it happens only in steps (b) and (e). But when these steps execute, all of the segment pins are OFF (because of steps (a) and (d)), so it should set both the first and second digits to all off (blank). Output is "  " rather than "89"!!

So it feels logically like it can't happen when the 8-segment pin-edges change, and it can't happen the 4-digit-selector-pin edges change. So, when can it happen??

Sorry to be so confused!

njackiw, 2018-05-29:
I've COMMITTED in the first draft of a new version of this, but it is silent only the question I have above, about how to actually program it for multiple digits.  If I come to understand this, I will modify the Description .docx further to explain it well to readers.

njackiw, 2018-05-29:
sorry I keep closing issues by accident! reopened since this is still for Leng to try to help me understand :-).

njackiw, 2018-05-29:
Ah, makes perfect sense! I should've thought of that. Thanks for the clear explanation -- will add an appropriate note to the Technical Background.