Alion3064492356/Sensors-for-RaspberryPi #28: Q: Also confused about digitalTempSensor

njackiw, 2018-05-28:
I have the same confusion -- or maybe Kuman has the same confusion? -- about **digitalTempSensor** as came up with REED SWITCHES (see #12) and HALL EFFECTS (see #10, now closed). In each of these cases, you have three technologies, one of which is clearly analog, one of which is clearly digital, and then one of which is some sort of hybrid that confuses me.

With temperature sensors:

**DS18B20 Temperature Sensor** is clearly a digital device. Description says "DS18B20 is a commonly used digital temperature sensor." No ADC is needed and the sensor seems to report via a 1-wire digital serial bus or port on Pin 7.

** analogTempSensor** is clearly an analog device, a thermistor. It reports a variable voltage (analog signal output) which gets fed into ADC where it can be read from ADC0382:CH0. 

So what is **digitalTempSensor**? It seems to be the same thing as **analogTempSensor**, which can be read directly from ADC0382:CH0, but which additionally includes an onboard trimPot for setting a threshold, and some additional circuitry that reports the analog temperature's relation to that threshold via a digital output ("HIGHer than threshold vs. LOWer than threshold") mapped to D0. 

Is that correct? If that's the case, first, I point out the sensor's name is wrong (#3 is not a digitalTempSensor; it's really an analogTempSensor plus some extra circuitry) and I struggle to understand why these devices even exist, since once i have the input from **analogTempSensor** digitized by ADC I can compare it to any threshold I want in software, without needing extra hardware.

However, one way that I _can_ make sense of this device is that it's _user-convenient_. It's convenient two ways. If I make something with it---let's say an airconditioner---and install it in someone's home, that person can adjust the threshold on which the device operates by playing with the trimmer pot rather than by editing software. (In this case, my device is using it as a digital threshold-exceeded-detector, with a user-provided threshhold rather than a software-defined threshhold.) Convenient for users! Or, if I'm a device-designer and don't want to configue a standalone ADC with every device, I can use this module because it already includes some sort of limited on-chip analog-to-digital conversion. (It won't tell me the actual temperature without an ADC to read A0, but it _will_ tell me whether that temperature exceeds the trimpot threshhold.)

Is this reading correct? If so, I propose we rename **35_ds18b20** to **35_digitalTempSensor** (which is a logical contrast to the current **analogTempSensor), and rename _this_ **28_digitalTempSensor** to **28_convenienceTempSensor.** Then, renumber activities so that new users encounter first the analog, then the digital, then the convenient tempSensor (which combines ideas from both), and in the **convenienceTempSensor** description, describe its purpose along the lines of the previous paragraph.

Let me know if I am understanding things correctly and you like this idea of the "convenienceTempSensor," in which case I think I know what to do, and can go back and change the one of the Hall Effect sensors to **convenienceHall** and the old **reedSwitch** to **convenienceReedSwitch**. If instead I"m misunderstanding things, please try to clarify for me!

Thanks.

Alion3064492356, 2018-05-28:
As for this problem, I agree with your proposal. It's not reasonable to name this module after digital temperature sensor. But we  need time to discuss the new name of this module.I wiil reply you tomorrow.

njackiw, 2018-05-28:
I hear from Emily that in general we have decided to keep the old module names (like magneticHall), but I will wait for your reply on this one, since here the name (original digitalTempSensor) isn't just unhelpful ... it's positively wrong (since it's really analog temperature sensor, with analog output and a convenient digital output). If you can comment on final name choice, I will wait on this one.

Alion3064492356, 2018-05-29:
Maybe we could name it after "AnalogTempSwitch".But please write some note in Read Me First.txt or description.doc to tell the revised reason. How do you feel like it?

Alion3064492356, 2018-05-29:
Analog temperature switch means that this module could be used as a temperature switch.

njackiw, 2018-05-30:
I understand the logic behind suggesting **switch**, which sounds more like a digital (or binary) comparison, but I also hear from Emily you are okay with **hybridTempSensor** which I think works better as a principle for naming many of these user-adjustable hybrid circuits, and for keeping people from confusing the analogTempSensor with the analogTempSwitch. (Especially since we use "Sensor" pretty loosely in Kuman kits! How is an LED a sensor?? :-)). 

I will try "hybridTempSensor" first. Please see my Overview description when I commit this one to see how I describe it and let me know if you want me to change it from that to "analogTempSwitch" or "analogTempSensorSwitch" (since it's really both).

njackiw, 2018-05-30:
If you are happy with the name "hybridTempSensor" after reviewing the above pull-request, you can close this Issue. Otherwise assign back to me!

njackiw, 2018-06-04:
Closing this one to clean up open Issue List. We have a bunch of issues that begin about a specific sensor, and wind up in a common conversation about what make good names, what are bad names, and how we handle this for users. We agree on the global strategy, so we no longer need all these original open Issues.

closed 2018-06-04 by njackiw