Alion3064492356/Sensors-for-RaspberryPi #32: opened for branch "AlionRevised2018/5/29"

Alion3064492356, 2018-05-29:
Hi Nick, I open this issue for us to discuss some problems included in branch "AlionRevised2018/5/29".
Please tell me the problems you find in this branch, and then I will correct it.
Thank you!

njackiw, 2018-05-29:
Wow!

First, great news about the decision to add resistors --- that's beautiful! Thanks!!

I have several comments and questions, so I will label all of the questions **Q1,** **Q2** etc. if that makes it easier to reply!

Really nice work here, Leng -- this will be a huge commit.

- Committed changes to 34_joystick are great .. I haven't touched that one yet, so will start from your changes after we've merged your branch.

- Committed changes to "update digital tube experiment" are going to conflict with my updates to some of the same files, which Emily has merged into master (#31) and which I've further modified in today's pull-request (#33). It looks to me like you've already merged my #31 changes but not yet seen my #33 changes. 

**Q1.** Can you handle these changes too when merging? They are both to 4bit-digital-tube ... the new pinout illustration inside the docx/Wiring Diagram, and the new docx/Technical Description section I added about controlling the multidigit display. If you'd prefer I (re)make my changes, let me know and I will do so Emily merges your commit. 

- You've pushed new Fritzing sensor labels throughout the activities and resistors into most of the LED experiments. Super!  But this hasn't been done entirely consistently.  In some places (like **TwoColorLed** you've updated the Fritzing diagrams and the DOCX/Experimental Materials; in other places (7ColorLED), just the Fritzing diagrams, etc. I think in order to add resistors consistently to an Experiment it affects a LOT of pieces:

A. Fritzing diagram, in *.jpg, *.cdr, and .docx/Wiring Diagram locations
B. Pin out diagram (add "(through resistor)" after the pin-to-pin mapping), in both "Pin position corresponding table.txt" and in copy/paste of that file into .docx/Wiring Diagram writeup.
C. Docx/Experimental Materials. Make sure ohm rating and quantity is listed here.
D. Docx/Experimental Procedure. Step 2 should change to say "...and use resistors and Dupont jumper wires as illustrated...", and in some LED 
activities, REMOVE the statement "(The XXXX LED module in this kit includes onboard series resistors, so no additional resistors are needed.)" since that turned out to be wrong.

**Q2.** This is a lot of changes---I am not listing them all!---so I suggest you do it where you can but don't worry about doing it more than for Step A (for graphic files) and C (for ohm rating); and that after your commit has been merged, I do a clean-up pass where I push the changes into all the other text files as necessary. I should do that only at a point where we aren't likely to be working in conflicting branches though. GIT is great at merging competing changes in text files semi-automatically, but no good at merging conflicting binary files (docx, jpg, cdr).

- I see you've made a lot of changes to filenames for consistency, better names, etc. This is usually a big improvement. But there are two places I wouldn't mind different names if they're possible:

**Q3.**  Is "TwoColorLed" a name we have to preserve because it's a conventional module name? If not, I prefer 2ColorLED for several reasons. Using digits instead of words (2 instead of "two") sorts better, makes the various LED modules look more like each other (e.g. 6ColorLED), and leads to shorter filenames; and uses the traditional English capitalization of LED which improves legibility. (LED is all-caps since it's an acronym.) So, I suggest we go back to 2ColorLED, unless you had a specific reason for intentionally switching to TwoColorLed.

**Q4. You've renamed **microphone** to **bigSoundSensor**. That's an improvement -- it's really a sensor, not a recorder -- and it fits better with **smallSoundSensor**, the other module. So I like it better than what we started. But it's still not very useful. Knowing they're big and small doesn't help us understand _why_ they're different, or even really _what_ they are except that they're different. You recently approved some name changes that will give us three temperature sensors named [analogTempSensor, digitalTempSensor, and hybridTempSensor]. Can we adopt those here too, and either have [smallSoundSensor, hybridSoundSensor] or better [analogSoundSensor, hybridSoundSensor]? I know Management's decided we should "keep the conventional names" most of the time, but I find relatively few Google hits (~1000) for "big sound sensor" compared to something like "mini reed switch" (~10,000)" or "hall magenetic" (60,000). So I don't know if the soundsensor names are conventional or if we're free to chose better ones...
 
 Okay, back to resistors:

**Q5.** Shouldn't **RGB_SMD_LED(3colorLed)** also now include resistors throughout the experiment?  My first tech support question to Kuman was about this module, and I was told I needed to add my own series resistors :-). If so, this implies changes A-D to the whole activity.

I think that's it. This is a huge improvement.  Can I ask you to work with Emily to try to merge it back to master as soon as you can? I'm hesistant to start English-work on any of the modules you've changed just because then we'll have conflicting branches. Feel free to go ahead and merge first even if you want to work on more fixes for **Q1 -- Q5** ... except for **Q2**, all of those fixes will just be in single experiments, so you can re-branch, fix just those specific activities, and recommit with much less opportunity for conflict.

Thanks Leng,
Nick

njackiw, 2018-05-29:
ps. Leng if you open another New Issue for my attention, please either ASSIGN it to me (top-right) or tag me in the comment with @njackiw. I won't see it by e-mail if neither of those things happen, and this one escaped me for a while til I stumbled over it in the big list of issues... Thanks!

Alion3064492356, 2018-05-30:
Thanks!
Q1:I have already merged it into master and  added  the first picture Emily sended to you yesterday.I added it because It would be helpful to customers to understand control principle.
Q2：I have dealt with these experiments according to Step B, Step C, Step D.
Q3:I have rename TwoColorLed to 2colorLED.
Q4:The difference between bigSoundSensor and smallSoundSensor is bigSoundSensor is much more sensitive to sound than smallSoundSensor.
We would better to try to name these modules refer to the product list. I am going to send you this list via email later. Customers might be confused about the differences between experiment name and name in list.
They may have a question"Which experiment is associated to the module in my hand?", and if there are very huge differences, they might feel annoyed.Even though we add some words to describe this difference in README.txt, we still worry about that not all customers would read this file at first. That is the key reason we insist not to change these names too much.
Q5:I have added resistors in RGB_SMD_LED(3colorLed). 
Thanks again!

closed 2018-05-30 by njackiw