Alion3064492356/Sensors-for-RaspberryPi #38: hybridTempSensor: please validate new Sample Code

njackiw, 2018-05-30:
In my reading of the original Sample Code for **hybridTempSensor** (formerly known as digitalTempSensor) code only printed anything if the digital threshold was exceeded. So no way for users to get feedback that the experiment was "working" unless they were over-threshold, which depending on the threshold, might not be the case.

I have changed the description of the intended outcome to:

> The Raspberry Pi command line output repeatedly outputs the numeric value of the analog temperature sensor.  Slowly approach the sensor with an open flame (a match or candle) to increase the local temperature. Once the sensor’s digital switch reports the increased heat has exceeded a certain level, the Raspberry Pi also reports “Temperature alarm! Threshold exceeded!”

I have modified the C and Python source code to, I think, do exactly this -- but I'm not able to test either of them since I don't have a Pi or the sensor.

Can you test my code works, or fix and recommit it if it doesn't?  If so, please add comments here about status and assign back to me. The new code still needs to be copy/pasted into the DOCX file once it works (which I can do). 

Thanks...

Alion3064492356, 2018-05-31:
OK, I'll validate this code.Thanks

Emily2333, 2018-06-01:
As for Question 38,as you and Leng checked above,Leng will test your code and update to you how it is,is it OK?

njackiw, 2018-06-04:
Yes, okay by me. This one is still on Leng's plate. Leng, if you've had a chance to test the code and it is good, please CLOSE this defect. Otherwise add comments about current state and assign to whoever should fix. Thanks.

closed 2018-06-25 by Alion3064492356