Alion3064492356/Sensors-for-RaspberryPi #41: Still need help with understanding Sound Sensors

njackiw, 2018-05-31:
So now we are calling them **bigSoundSensor** and **smallSoundSensor**.  Somewhere Leng has told me the first one is for "bigger sounds."

Q: What is a "big" sound and what is a "small" sound?? How do I explain these as meaningfully different to users?

Both sensors look the same, have the same wiring diagrams, and have identical source code (except in one, the input pin is named MIC_D0_PIN and in the other SOUND_D0_PIN). For both, the sample code makes it clear that "voice" is a good example of both a BigSound and a SmallSound. Both have threshhold adjusters which make it seem like they're good for a range of sound volumes, not just "one" volume.

I've tried to watch https://www.youtube.com/watch?v=l-H0KVFfkgs which seems to demonstrate the difference, but it's in Spanish and if he's actually saying anything dfifferent from "one is larger than the other" I can't tell what it is. Otherwise I can find no difference between these...

If there is no difference between them other than physical dimensions of the modules (size of the "sensors", rather than size of the "sounds"), I think we should collapse them into one folder, with one set of documents, and inside the *.DOCX there simply list it as available in two sizes. This is what we do with "Two-Color LED (3mm or 5mm)".

Is there a difference?

Emily2333, 2018-06-01:
The difference between bigSoundSensor and smallSoundSensor is the sensibility of sound.The bigSoundSensor's is more sensitive to sound than smallSoundSensor.That means the smallest or lowest voice which bigSoundSensor sensed is smaller than which smallSoundSensor did.

njackiw, 2018-06-01:
so, to repeat, your'e saying the _big_ sound sensor detects _smaller_ sounds than the _small_ sound sensor?  ha ha okay.  that's backwards ... but that's what we get with "conventional names." It's really the _sensor_, not the _sound_, which is big vs. small! Oh well... 

I will try to make their similarity and difference clear in the two experiments.