Alion3064492356/Sensors-for-RaspberryPi #42: Question/TiltSwitch: code vs. Description

njackiw, 2018-05-31:
Back to TiltSwitch. (Sorry, I had thought I was "done" with issues but apparently there are still many more!)

The .docx file says

> When the tilt switch is in the horizontal position, the S pin of the switch module is disconnected from the GND pin,and the S pin is connected to VCC through the pull-up resistor, the level on the S pin is at high level. When the tilt switch is tilted at a certain angle, the internal metal balls will roll, and then the S pin and the GND pin are connected through the metal ball. At this time, the level of the S pin becomes a low level. 

So this suggests (a) TILTED=LOW, HORIZONTAL=HIGH.

But the C code says:
```
		if(0 == digitalRead(TiltPin))
		{
			digitalWrite(LedPin, LOW);	
		}	
		else
		{
			digitalWrite(LedPin, HIGH);	
		}

```
and the **Experimental phenomena** says:

> When the tilt switch is tilted at a certain angle, the LED lamp is lit, and when the tilt switch is in the horizontal position, the light will be not lit.

So this suggests (b) TILTED=HIGH (LED on), HORIZONTAL=LOW (LED off).

Which is it? (a) or (b)?

njackiw, 2018-05-31:
Also, separate question, do the Python and C samples actually do the same thing? Just reading them, it looks to me like the C code mirrors the state of the switch in the state of the LED. Switch open = LED open. But the Python seems to SWITCH the LED when the switch state GPIO.FALLING. If it only falls when the switch closes, then half the time you close the switch, you will turn the LED off (like the C code), but half the time you close the switch, you will turn the LED on (unlike the C code). And opening the switch won't do anything (unlike the C code).

If I'm right about this, can we change the Python code to match the C code? It's a simpler program and a more straightforward picture for users to understand.

So, two questions:
Q1. What is the TiltSwitch output when the switch is OPEN (Horizontal)?
Q2. Should the Python change to match the functionality of the C?

Alion3064492356, 2018-06-01:
Thank you for the suggestions mentioned above.I have made corrections on the basis of your advice and update the branch "master". It would be great if you could recheck this correction.

Alion3064492356, 2018-06-01:
Both Q1 and Q2 have already been done.Thanks.

PR #46 (2018-06-02): NickFriday: new: tracking, touch, and sound x2; also lots of cleanup
-trackingSensor: first English version

-touchSensor: first English version (but still open issue #45: inconsistent sample code)

-bigSoundSensor, smallSoundSensor: first English revision, keeping these two sensors as separate folders with “conventional” sensor names, while making it clear that their source code, wiring diagram, and experimental procedures were identical. I also made the source code files ACTUALLY identical, and renamed them to reflect that identity.  Fixing #41.

-obstacleAvoidance: fixed #43  so pinout tabl [... cut]
closed 2018-06-01 by njackiw