Alion3064492356/Sensors-for-RaspberryPi #43: Q: Obstacle Avoidance/questions on pins and pots

njackiw, 2018-05-31:
Two questions on **Obstacle Avoidance Sensor**:

Q1. The module photograph and wiring diagram shows a sensor with four output pins labeled GND, +, OUT, EN.  But the Pin position table talks about only three pins, "S", "-", and "+".  Our docs should call them by the same name as on the module.  Can you confirm "S" should be named "OUT" instead? And what is "EN?"

Q2. There is no mention in the original activity of the fact there are TWO pots on the board. Can you confirm the technical accuracy of the following paragraph I'd like to add?

>You can adjust the sensitivity of its distance detector by adjusting the lefthand onboard potentiometer, just above the sensor’s GND pin. The effective sensing range is ~2cm to 40cm. (The righthand onboard potentiometer, above the EN pin, controls the frequency of the infrared transmitter. You will likely have no need to adjust this.)

njackiw, 2018-06-01:
Correction to Q2 -- I found a website with a sensor closer to Kuman's. Here's what I think the paragraph should say now. Can you confirm technical accuracy, or correct?

> You can adjust the sensitivity of its distance detector by adjusting the righthand onboard potentiometer, labeled “205” just above the sensor’s EN pin. The effective sensing range is ~2cm to 40cm. (The righthand onboard potentiometer, labeled “103” above the GND pin, controls the frequency of the infrared transmitter. You will likely have no need to adjust this.)

Alion3064492356, 2018-06-01:
I think "S" should be named "OUT" in Pin position corresponding table.txt. It's a good revision you mentioned above.
As for the pin "EN", this pin "EN"  is used to enable or disable this module.When the level of pin EN  is high, this module is enabled.Or when it turns to low level, this module is disabled. Because pin EN is pulled up to VCC through 22KΩ resistor, it's default level is high.As a result, this module is enabled when pin EN is connected to nothing.That's why we didn't connect pin En .

Alion3064492356, 2018-06-01:
Q2: It‘s better to write the description you mentioned above. Thank you!

PR #46 (2018-06-02): NickFriday: new: tracking, touch, and sound x2; also lots of cleanup
-trackingSensor: first English version

-touchSensor: first English version (but still open issue #45: inconsistent sample code)

-bigSoundSensor, smallSoundSensor: first English revision, keeping these two sensors as separate folders with “conventional” sensor names, while making it clear that their source code, wiring diagram, and experimental procedures were identical. I also made the source code files ACTUALLY identical, and renamed them to reflect that identity.  Fixing #41.

-obstacleAvoidance: fixed #43  so pinout tabl [... cut]
closed 2018-06-01 by njackiw