Alion3064492356/Sensors-for-RaspberryPi #45: TouchSensor/inconsistent sample code

njackiw, 2018-06-01:
TouchSensor Experiment says
> "1.	When you touch the sensor, the LED illuminates. When you stop touching the sensor, it goes dark."
(Those are my words, but Original text said something similar.)

Python code does exactly what this says:
```
	while True:
		if GPIO.input(TouchPin) == GPIO.HIGH:
			print 'touched!'
			GPIO.output(LedPin, GPIO.HIGH)  # led on
                        time.sleep(0.2)
		else:	
			GPIO.output(LedPin, GPIO.LOW) # led off
```
In other words, high input drives high output.  But C code does something different:

```
if(digitalRead(TouchPin) == HIGH)
		{
			printf("touched\n");
			digitalWrite(LedPin, HIGH);     //led on
			delay(100);
			digitalWrite(LedPin, LOW);    //led off
		}	

```
In other words, high code _strobes_ high output, but leaves it low.  If high input never arises, output is undefined -- might be high, or low, depending on ambient conditions.

I propose we replace the C loop body with

`digitalWrite(LedPin,` digitalRead(TouchPin)); // Set LED high when Touch is high; or low when low

This seems to me both correct, and simpler.

If you agree, please change both **touchSensor.c** and ***.docx/Sample Code/C Code** (or tell me to).

Alion3064492356, 2018-06-02:
Okay, I am gonna change both these files right now.

njackiw, 2018-06-03:
Leng fixed this and I've verified the fix; so I'm closing it.

closed 2018-06-03 by njackiw