Alion3064492356/Sensors-for-RaspberryPi #47: flameSensor: sample code issue

njackiw, 2018-06-02:
**flameSensor** follows the typical hybrid design: A0 reports analog voltage through ADC; D0 reports whether that voltage exceeds a user-controlled, trimPot-determined threshold.

However, unlike other hybrid sensor source code, the source code in **flameSensor** reports TWO different thresholds --- one from D0 (which is set by the hardware trimPot), and one by comparing the analog signal to a software constant.

Here it is in Python:
```
        if(digitalVal == 1):                                          ## report the D0 hardware threshold by text
            print 'DO is %d' % digitalVal  
...
	    thresholdVal = 150

	    if(analogVal < thresholdVal):
		GPIO.output(LedPin, GPIO.HIGH)     ## report the A0 software threshold by LED

```
Then here it is in C:
```
		if((digitalVal = digitalRead(Flame_DO_Pin)))
			printf("Do is %d.\n", digitalVal);   // report D0 hardware threshold as text
...

		#define thresholdVal	150
		if(analogVal < thresholdVal)
				digitalWrite(LedPin, HIGH);   // report the A0 software threshold by LED
```

I think these two different thresholds are confusing to users. In general, in our hybrid experiments, we've explained the threshold as what you set by the trimPot, rather than as a software constant. I would like to simplify these to something more like this (pseudo-code)

```
while (1)
{
               print("A0 is ", getADCvalue());   // report A0 to command line interface
              digitalWrite(LedPin, digitalRead(flameD0pin));  // digital switch to LED
}
```

This makes it clear the sensor hardware itself only implements one threshold (the pot); and it makes the experiment report the hybrid's two sensors in my suggested "typeOf(input)=typeOf(output)" experimental design as per #36.  In other words, the analog A0 gets reported by a (varying) real number; the digital D0 gets reported by an on/off LED.

If you like this proposed redesign, can you write and test the appropriate code from my pseudocode?

If you don't like this proposed redesign, is there perhaps an explanation for why this experiment has the second (software-only) threshold that I am not understanding?

Alion3064492356, 2018-06-02:
Ok, thanks! I prefer your suggestion, and I will revise it and commit into branch "master".

njackiw, 2018-06-02:
looks good. thanks and closed!

closed 2018-06-02 by njackiw