Alion3064492356/Sensors-for-RaspberryPi #48: joyStick: proposed changes to sample code

njackiw, 2018-06-02:
Joystick reports X-axis, Y-axis, and button separately on three different pins.  This means it can probably report three variables simultaneously -- in other words, its "state"  is something like  [LEFT, UP, PRESSED]  or [RIGHT, UP, NOT PRESSED].

But the current sample code, though, both Python and C, reports at most ONE input variable at a time. Consider Python:

```
def getResult():										#get joystick result
	if ADC0832.getResult(1) == 0:
		return 1		#up
	if ADC0832.getResult(1) == 255:
		return 2		#down

	if ADC0832.getResult(0) == 0:
		return 3		#left
	if ADC0832.getResult(0) == 255:
		return 4		#right

	if GPIO.input(btn) == 0:
		print 'Button is pressed!'						# Button pressed
```

If they're pressing y-axis up, ADC0832.getResult(1)==0  will return immediately, and we'll never "see" that they were also pressing x-LEFT and BUTTON-DOWN.

C code is a little better. It will report the BUTTON value regardless of the x/y values. But it will report at most X **or** Y, never both, because tmp can only have one value at the end of this code:

```
		xVal = get_ADC_Result('x');
		if(xVal == 0)
		{
			tmp = 1; //up	
		}
		if(xVal == 255)
		{
			tmp = 2; //down
		}

		yVal = get_ADC_Result('y');
		if(yVal == 0)
		{
			tmp = 3; //left
		}
		if(yVal == 255)
		{
			tmp = 4; //right
		}

		zVal = digitalRead(JoyStick_Z);
		if(zVal == 0)
		{
			printf("Button is pressed !\n");
		}

```

I would prefer simpler sample code that demonstrated that the joystick is capable of three inputs _simultaneously_, rather than "one at a time." 

Also, there are two other conceptual problems here:

- I think the button should **not** be named _JoyStick_Z_, which suggests the z-axis (x, y, z). Buttons have nothing to do with the z-axis or the "joystick" directions. Let's call it _buttonValue_.

- The above code comments say yVal decides LEFT/RIGHT (horizontal directions) and xVal decides UP/DOWN (vertical directions). But in mathematics, x is horizontal and y is vertical. So these interpretations should be switched.   I think the C code is just wrong here, and that getADCValue('x') should decide left/right, not up/down.

In C-pseudo code (untested because I don't have a Raspberry Pi and the joystick), I think my proposed code responding to these looks something like this:
```
while (1)
{
		// read the three inputs
		xVal = get_ADC_Result('x');
                yVal = get_ADC_Result('y');
                buttonValue = digitalRead(buttonPin);

		// diagnose them on the command-line
		if(xVal == 0)
                        printf("left ");
                else if (xVal == 255)
                         printf("right ");

		if(yVal == 0)
                        printf("up ");
                else if (xVal == 255)
                         printf("down ");

                if (buttonValue == HIGH)
                          printf("pressed"); /// else: not pressed!

              // and if we generated any output above, move to a new line
               if ((HIGH == buttonValue) || (((0 == xVal) || (255 == xVal)) || ((0 == yVal) || (255 == yVal)))
                         printf("\n");
}
```

I think this should be able to report all three inputs simultaneously, and is a bit easier to follow (except perhaps the last line). Do you agree? And does it work? If so, can you change the Python to match?

Alion3064492356, 2018-06-02:
I have done this work ,tested the code and committed.

njackiw, 2018-06-02:
looks good! I changed "zval" to "bVal" for the button, just to take it out of the spatial 3D namespace of (x, y, z).

Closing with joystick checkin.

closed 2018-06-02 by njackiw