Alion3064492356/Sensors-for-RaspberryPi #50: relay: more sample code change requests

njackiw, 2018-06-03:
I think "relay" is hard for people to understand if they are new to electronics, because it doesn't seem to >do< anything they can't already do without a relay?

> aa_LED turns an LED on and off.
> zz_RELAY turns an LED on and off. 
> "So," they say... "What's the difference?"
> 

I would like to make the sample code as easy to understand as possible, and perhaps change the DOCX a bit more to talk about what the sample code is doing. Right now, though, the sample code wires the LED through pin 16 (and a resistor), and code turns on the LED (digitalWrite(ledPin, HIGH)) before entering the loop that blinks the relay.

I wonder instead if we could change it to not use Pin 16 and not use ledPin at all in the source code. Just wire  +5V through the resistor to LED through relay to GND. Then in DOCX I can say something like

> Without the relay in the loop, this _working circuit_ would create an LED that was always on: it connects power to ground.  The relay is introduced as a switch into the _working circuit_ and the _control circuit_ controls the relay through Pin 10. Notice that the controlling source code does not control the LED itself; it only controls the relay. As the source code signals the relay, the relay switch opens and closes the working circuit, causing the LED to blink.
 
Does this seem like a conceptual improvement? And if so, is it worth the cost of you changing the C, and Python, and wiring diagram in .jpg/.cdr/.docx and pin outs?   Sorry :-).

Alion3064492356, 2018-06-04:
It's a good idea. It would be more reasonable and I will improve this experiment on the basis of your advice.
For safety, I think we could connect LED to +3.3V through resistor.

njackiw, 2018-06-04:
Thanks, you're fast!

Added explanations to DOCX/Steps 2 & 4 to explain new code.

closed 2018-06-04 by njackiw