Alion3064492356/Sensors-for-RaspberryPi #57: opened for some problems in description.doc

Alion3064492356, 2018-06-08:
1.Hi,Nick! I have a proposal about lightblock.doc that we'd better to add a picture to explain the principle of lightblock module. I feel this picture is helpful when we try to understand lightblock module.
Note: I cannot add this picture into this issue for some unknown reason, as a result ,I am going to add it into  
the lightblock.doc. If you feel it's unnecessary, I could delete it.

Alion3064492356, 2018-06-08:
2.Hi,Nick! A proposal about smallSoundSensor.doc  and bigSoundSensor.doc that we'd better to add a picture to explain the principle of soundSensor module. I feel this picture is helpful when we try to understand soundSensor module.
Note: I cannot add this picture into this issue for some unknown reason, as a result ,I am going to add it into the soundSensor.doc. If you feel it's unnecessary, I could delete it.

Alion3064492356, 2018-06-08:
3. Hi, Nick!  I revised the reedSwitch experiment. I wish you to recheck this experiment, and please inform me     
if you find any problems.

njackiw, 2018-06-08:
Hi Leng,

I've checked these out, thanks. Feedback:

1. I like the illustration in the lightblock.doc; it feels to me appropriate for the audience, who I imagine to be relatively new to schematic diagrams but can probably recognize a diode and a switch. Let's keep it! I've added a bit more explanation to the diagram there I will commit soon. 

2. I don't like the diagram in bigsoundsensor.doc and smallsoundsensor.doc, though.  Knowing there is an "internal frontal shell" and an "insulating mat" and a "metal plate" doesn't help me make sense of how it works, and these words all mean not just different things, but different TYPES of things ("internal" is a type of _location_; "insulating" is a type of _function_; "metal" is an type of _ingredient_) so make me ask things like "what is the _function_ of the metal plate?" and "what is the _composition of the shell? " which the diagram does not answer.   Also, "field-effect transistor" is misspelled (it says "filed-effect", not "field-effect;" "filed" IS an English word so it's not clear to a new user that this is a spelling error.)  In summary, I think the illustration adds more problems than it solves and would recommend removing it.
 
3. I look at "reedswitch.docx" and am not sure I understand the new experiment. It looks to me like you've removed the ADC entirely from the experiment. What we are left with has some possible problems, some of which may just be my misunderstandings, but let me list them all:

a. The pinouts now reference "Raspberry Pi +3.3V". Every other sensor is using the +5V power tap. Is this one really supposed to be different, or are all the other ones wrong?

b. The source code now refers to ANALOG_PIN_A0 but uses digitalRead() without going through the ADC. Is this really the right name or approach? In all other "hybrid" experiment designs, the analogPin goes to ADC and the digital pin goes straight to digitalRead(). I understand the "analog pin" here is really reporting on a fundamentally non-analog phenomenon (switch open/closed), but still think for clarity if we want to go straight to digitalRead(), we should be reading the digital pin D0 rather than the analog one. 

c. Just at the formatting level, I want to make a few changes to the Pin Outs  (*.docx, *.txt) after you are done with them to keep them consistent with other activities.  I can do this if you can sign off on (a) and (b) as correct, or let me know when you are done with further edits you decide to make.

Alion3064492356, 2018-06-09:
Hi Nick,
Thanks!
2.To be honest, I am not clear about the soundsensor structure and I copied this picture from Internet.
Now that this picture has these error, we can delete it. What do you think?
3.This is a big problem that would confuse customer. 
a. I reconnect pin "+" to "+3.3" because I think it would be more safe to RaspberryPi. 
Because the AO pin of this module is directly connected to the Raspberry Pi GPIO without through the ADC, and the resistance value between the AO pin and VCC is relatively small, so VCC is changed to +3.3. And other similar modules AO is connected to the ADC chip, so don't worry about voltage issues
b.I tested the level of AO and found that when the magnet is close to the reed switch, the level of AO drops from 3.3V to about 0V. When the magnet is far away, the level of AO goes from 0V to 3.3V.It turns out that the switch is digital , and the AO pin also outputs the digital signal instead of the analog signal. Therefore, the AO signal can be directly sent to the Raspberry Pi, and the actual test results are also normal.  If we should explain this feature in reedSwitch.doc or not?

njackiw, 2018-06-10:
2. Agreed; let's get rid of this picture.
3. Ok, this makes some sense to me, but maybe better if you try to explain it more fully in reedSwitch than if I do. I can edit your language after if you want.

Alion3064492356, 2018-06-11:
Hi Nick,
I went though the "introduction to sensors.doc", and I found some problems.
1. In the third paragraph, you said " “Maximum” voltage would indicate a very hot temperature.", I feel this conclusion is not inalterable, and It's related to the characteristics of the thermistor and circuit design. In the same circuit structure, the NTC thermistor and PTC thermistor will result different conclusions.
Thanks Nick!

njackiw, 2018-06-11:
Okay, I see your point. But I am describing an IMAGINARY analog thermometer, not a real one, so I want to keep it simple and clear as a thought experiment—after all, I'm trying to explain analog vs. digital here, not thermometers! 

Do you think it would be better if these two sentences said

"In an analog design, this sensor might report the current temperature as an electrical voltage. The designer, or manufacturer, could choose no voltage—or very low voltage—on the wire to indicate a very cold temperature. And some appropriate “maximum” voltage for the circuit —  perhaps 3.3V — might be used to indicate a very hot temperature.”

Perhaps that phrasing makes it sound more like a “thought experiment” rather than a description of a specific analog sensor.

Also you said “I found some problems” — problems plural. Were there other parts you thought should be clearer or better said? I totally welcome your feedback!

Thanks Leng,
Nick

Alion3064492356, 2018-06-11:
Hi Nick,
I agree to replace these two sentences. And I didn't find more problems in this file.
Why I wrote  "some problems" ?
That is a mistake. I wrote that commit as soon as I found this problem mentioned in that commit, and I thought there might be some other problems, so I wrote "some problems".However, I didn't find any other problem in the end, but I forgot to correct this words.
Thanks for your efforts!
Leng

njackiw, 2018-06-13:
This has been fixed so I am closing...

closed 2018-06-13 by njackiw