Alion3064492356/Sensors-for-RaspberryPi #60: smokeSensor sample code and description is weak

njackiw, 2018-06-13:
The smokeSensor seems to me to be a hybrid circuit design, reporting a digital smoke intensity on A0 and "something else" on digital D0.  In all likelihood, what gets reported on D0 is whether the A0 signal has exceeded a user-adjustable threshold determined by a trimPot ... this is how all the other hybrid designs seem to work, and web-searching for generic MQ2s from other manufacturers suggest most of them are designed like this. 

However, the original Experiment doesn't ever mention D0 or what gets reported there, and the original sample code doesn't make any use of D0, and there is no photograph or mention in the original Experiment of an onboard trimPot so I don't know whether it's really like this or not!

I have left the description describing the original experimental source code, which ignores D0 and tests A0 against a threshold hard-coded in the Source Code rather than user-adjustable.  But if you have time, it would be really nice to change this source code (and perhaps even me update the Experiment text) to be consistent with our other Source Codes for hybrid sensor designs, which schematically should look like this:

while (1)
{
        print(A0_pin);    // report ANALOG value to the console 
        digitalWrite(LED_pin, digitalRead(D0_pin));  // report DIGITAL threshold both by LED
        if (digitalRead(D0_pin))
              print("trimpot threshold exceeded!")     // and to the console
        delay(1.0);
}

Then they can vary the threshold with the trimmer, rather than by changing the source code and recompiling.

Assigning to Leng for consideration. The current code and description are shippable if we are in a hurry to complete today, but if you want to revise this, please update the source code and assign back to me and I'll update the description to match by tomorrow.

Alion3064492356, 2018-06-14:
Hi, Nick,
Thanks for your suggestion. I have adjusted the code in smokeSensor and uploaded to the branch master. Hope you to validate it and update the description. 
Thanks!
Leng

Alion3064492356, 2018-06-14:
Hi Nick,
Thanks for your suggestion. I have adjusted the code in expriment smokeSensor and modified the description on the base of the one you updated yesterday. So there is no need to do more work on this experiment.
Thanks again!
Leng

closed 2018-06-25 by Alion3064492356